Aceinna/docs_aceinna-dmu380 #7: How to differentiate on unit via SPI interface

yifanff, 2019-07-15:
As for openimu300 and openimu330 unit, How to differentiate units via SPI interface?  though  modelstring could be read through UART interface to differentiate units, It seems no unit ID could be read through SPI interface.

AndreyBondarev, 2019-07-15:
Table1 in the SPI engine description updated to include corresponding registers. Also since it's open source customers may implement their own mechanism of unit identification

yifanff, 2019-07-16:
Ok, thanks. In the updated register map, why there are two OpenIMU330 words left?  what is the default Product ID with factory released FW currently?

AndreyBondarev, 2019-07-16:
Thanks for the good catch - fixed

yifanff, 2019-07-25:
Ok，thanks，what is default product ID via SPI interface currently?

AndreyBondarev, 2019-07-25:
Default ID on OpenIMU300 is 0x3000 and on OpenIMU330 0x3300 (on SPI interface)

closed 2019-07-28 by yifanff