Aceinna/docs_aceinna-dmu380 #19: 330SPI--time sequence between master starting time and 330 power on time

erkuo521, 2019-11-13:
we should highlight 330 SPI items in our on-line doc:
- master starting time firstly, and then 330 power on. or the SPI will see messy code, and i have to restart 330 again.

-like 381, 0x34/0x35 self_test register and function also need to realized, detection result in status 0x3C

- DRDY need to keep same with 381

AndreyBondarev, 2019-12-17:
Startup sequence already provided in p.16 here: 
https://openimu.readthedocs.io/en/latest/software/SPImessaging.html
So far there was no plan to implement Self test commands on SPI interface.
In new OpenIMU330/300 DRDY active edge is the falling - same as in applications for JD.
And it's described on the same page as above

erkuo521, 2019-12-18:
thanks a lot for your so detailed explain in doc

closed 2020-02-05 by m5horton