Aceinna/docs_aceinna-dmu380 #22: add pps function description

yifanff, 2019-12-11:


joem-at-aceinna, 2019-12-18:
Generated an initial draft of section on the read-the-docs site that describe the synchronization process.  Awaiting feedback on this topic.  Once I incorporate the feedback I will close the task.

yifanff, 2019-12-20:
customer need pps function description on both  OpenIMU300 and OpenIMU330

yifanff, 2020-03-16:
PPS application note in document should be update with the items below:
1. The PPS 1Hz signal high voltage duty cycle supported.(0.01% From 1 second interval is 100 microseconds)
2. Which kind of GNSS receiver supported.
3. The typical setup time when PPS raising edge locked with data ready signal raising edge.
4. The stable time inteval between PPS raising edge and data ready ready signal raising edge.
5. The PPS timestamp desription. 
6.The different API between OpenIMU330BI and OpenIMU300ZI in open source code should be clearify.