Aceinna/docs_aceinna-dmu380 #25: 300/330 orientation; replace 381 pic in doc; indicate the default value of LPF in SPI

erkuo521, 2019-12-18:
Comments on subject mentioned topic by pics:
[image]

[image]

erkuo521, 2019-12-18:
Note 3: Magnetometer sensors scale is **16354** LSB/Gauss, the value should be **16384**. 
the unit of time in table 12 of SPI should be **uSec**, not **nSec**:

tDELAY | Time between successiveclock cycles | 9 (min) | 16 | **nSec**
-- | -- | -- | -- | --

yifanff, 2020-01-09:
as for the 330 special default orientation(uart/spi interface), please update the info in the document.